dstroy0/InputHandler #2: example formatting

dstroy0, 2022-03-01:
Examples do not follow arduino formatting conventions.

dstroy0, 2022-03-02:
added .clang-format to library, the style is not right for the examples yet.  Arduino uses an artistic style to format where the closing braces are on a newline.

dstroy0, 2022-03-02:
Examples follow Arduino IDE formatting conventions, but I would like to make the C++ extension's clang-format in vscode behave the same way for the examples.

2bndy5, 2022-03-03:
> clang-format for examples still isn't right

The [Arduino CLI workflow is using the examples_formatter.conf file](https://github.com/dstroy0/InputHandler/blob/f6bdcfd6640afa4e37cf07397dbff6f055405bcf/.github/workflows/build_arduino_cli.yml#L14) in your examples folder. This conf file (& the action that uses it) executes a different linter called AStyle.

Arduino IDE is moving away from AStyle because it has stopped seeing updates (including possible bug fixes). Going forward, the Arduino IDE will start using clang-format, but we won't see that until the Arduino IDE v2.0 reaches stable release.

I received a copy of the proposed Arduino IDE .clang-format file from someone on the Arduino team (with admin privileges). You can get [a copy from the RF24/clang-format branch](https://github.com/nRF24/RF24/blob/clang-format/examples/.clang-format).

To actually use clang-format as a linter in you CI workflows, I helped write [a Github Action that you can use](https://github.com/shenxianpeng/cpp-linter-action#readme) like so:
```yml
      - uses: shenxianpeng/cpp-linter-action@master
        id: linter
        with:
          version: 12
          extensions: ino
          style: file
          tidy-checks: '-*'
          files-changed-only: false
          thread-comments: false
          ignore: examples/old_backups
          verbosity: 20

      - name: Linter checks failed?
        if: steps.linter.outputs.checks-failed > 0
        run: exit 1
```

dstroy0, 2022-03-05:
Excellent, I've moved .clang-format to src/utility.  I made the lib-cpp-linter workflow using ```shenxianpeng/cpp-linter-action@master```

closed 2022-03-05 by dstroy0