dstroy0/InputHandler #8: Could a integer be used as a UI cmd?

2bndy5, 2022-03-03:
Following the convention laid out by the Firmata library, a controlling input (albeit a user or external program) can specify a different input command using a single unsigned char. Is this something that could be extended to this lib?

More specifically, if this is possible, I would like to not require a null byte terminator for this scenario. Thus, the title says integer, not encoded c-string.

dstroy0, 2022-03-04:
At its heart it is using strcmp to compare the first position input token the command list built by UserCommandParameters.  Any unsigned char should trigger commands.  I have a few sensors that send out serial strings and I've been experimenting using this library to parse their output which looks like  "H 12345 T 12345 Z 00010\r\n"  so I made a command that triggers on H and looks for those other tokens.  In theory ANY unsigned char should work.

2bndy5, 2022-03-04:
>  I have a few sensors that send out serial strings

Holy crap! This could be used as a GPS parser!

I'll have to spin up a modified example and try sending hex digits over the serial monitor.

dstroy0, 2022-03-12:
This functionality is available.

closed 2022-03-12 by dstroy0