dstroy0/InputHandler #11: add optional max N_ARGS to `UserCommandParameters()`

2bndy5, 2022-03-04:
This is more of a long term goal, but I thought it would be useful to let users set a minimum required number of args (per command) and an optional maximum number of args. This would kind of follow C++ conventions about optional function args, but there would have to be a way to determine if the optional args are present before attempting to extract/interpret them from the input.

To be clear, the existing `number_of_arguments` parameter is already treated like the required minimum .

Is this related to (or maybe a duplicate of) #10?

2bndy5, 2022-03-04:
This would also imply that the length of the `argument_type_array` would come **after** the that parameter (which complies with buffer length convention in function signatures).

https://github.com/dstroy0/InputHandler/blob/f6bdcfd6640afa4e37cf07397dbff6f055405bcf/InputHandler.h#L117-L120

would be changed to
```cpp
    UserCommandParameters(const char* user_defined_command_to_match,
                          void (*user_defined_function_to_call)(UserInput*),
                          const UITYPE argument_type_array[] = NULL,
                          size_t number_of_arguments = 0,
                          size_t max_number_of_arguments = 0)
```

dstroy0, 2022-03-04:
Yes this is very feasible.

2bndy5, 2022-03-06:
> Do you want to be able to pass a variadic number of arguments in a predefined range? I think you mentioned right now num_args functions as `max_num_args` I can add a `min_num_args` with a default of `max_num_args`. Then you can change `min_num_args` and it will accept args in the range of ` num_args >= min_num_args && num_args <= max_num_args`

My idea here is a user could
```cpp
const UITYPE arg_pattern = { UITYPE::UINT8_T, INT_8T };
UserCommandParameters test_cmd(
  "test",
  cmd_parser_callback,
  arg_pattern,
  1, // the min number of REQUIRED args in the designated pattern
  2  // the max number of args in the designated pattern
);
```
If the second "max_number_of_args" integer is not given, then the minimum number of args should match the length of the `arg_pattern` array because they are all required args (not optional).

dstroy0, 2022-03-08:
We can start work on this now.

dstroy0, 2022-03-08:
feature implemented

closed 2022-03-08 by dstroy0