dstroy0/InputHandler #13: username only required for debug output

2bndy5, 2022-03-04:
It could be removed from the c'tor and declared like
```cpp
#ifdef (_DEBUG_USER_INPUT)
    #define _USERNAME_ "user"
#endif
```

Or just not use it at all since we're not tracking multiple users' characteristic data. I wish there was a way to label a stream obj and use that instead of a username. Or maybe just differentiate between a buffer vs `Stream` operation.

dstroy0, 2022-03-04:
Yes it’s there to help mostly.  I use the mDNS host name as my “username” so I can swap between loads of microcontrollers and know I’m entering commands into the right one.  I also thought about allowing it to be swapped on the fly so if someone else logs their name gets echoed in the log.  To help audit.  More than one person can use some of the equipment.  I think we can make it default NULL.

2bndy5, 2022-03-05:
Ah, now I see - its log msg origin ID. We attempted to make a RF24Log library that can be used on any platform (even MacOS or Windows). Development halted when we wanted to have an identifier that indicated the origin of the logging msg. Since it was originally designed to have a small footprint, we couldn't decide how to delegate the origin's ID (& corresponding log level) without allocating more memory...

In the meantime, you could
```cpp
#ifdef (_DEBUG_USER_INPUT)
    #ifndef (LOGGER_ID)
        #define LOGGER_ID "user"
    #endif
#endif
```
In platformIO, this could easily be set per device in the platformio.ini
```ini
[env:device1]
board = "esp32"
build_flags =
    -DLOGGER_ID="device1"

[env:device2]
board = "esp8266"
build_flags =
    -DLOGGER_ID="device2"
```

dstroy0, 2022-03-05:
Would leaving it as default NULL and moving it to the end of the initialization list be a good compromise?

2bndy5, 2022-03-05:
I suppose if you're using the arduino IDE then you'd have no easy way to dynamically change it.

I'm ok with leaving it as is. But, for library debut release, it is a buffer of bytes that doesn't need to occupy memory unless debugging is enabled. Leave this issue open as reminder.

I've been taking a closer look at what could be done to finish RF24Log, but I can't recommend using an API that is likely to change.

dstroy0, 2022-03-12:
This is fixed, it is a const char* with a default of "".

closed 2022-03-12 by dstroy0