dstroy0/InputHandler #22: documentation

dstroy0, 2022-03-17:
I'm having trouble with doxygen documenting the members of the UserInput Constants section.  Doxygen installed locally reproduces the error so I think it is either my documentation syntax or my configuration.

2bndy5, 2022-03-17:
Looking at the latest build log, I see the following warnings:
```
/home/runner/work/InputHandler/InputHandler/src/InputHandler.h:57: warning: documented symbol 'ui_input_type_strings_pgm' was not declared or defined.
/home/runner/work/InputHandler/InputHandler/src/InputHandler.h:84: warning: documented symbol 'const char ui_escaped_char_pgm' was not declared or defined.
```

I'm pretty sure the awesome doxygen theme requires doxygen v1.9.1 or newer. Generated docs' footer says v1.8.17 was used.

Lastly, doxygen has problems when `PROGMEM` comes after the object name. It should go between the modifiers and data type for Doxygen to parse the code properly. 
```cpp
const PROGMEM char *name = "name";
```
According to the Arduino docs, putting `PROGMEM` after the object name is only for backward compatibility with extremely old versions of Arduino IDE (back before they started using a `v` to prefix the version number - some time before 2015).

dstroy0, 2022-03-17:
I saw that as well, I tried adding PROGMEM to PREDEFINED.  I’ll just change the order of the attribute like you said.   I’m using doxygen 1.9.3 local, I recently updated the workflow because doxygen-awesome was updated to work with 1.9.3 and I was using 1.9.2 before.

dstroy0, 2022-03-17:
This is fixed, gh-pages are also deploying automatically again.

closed 2022-03-17 by dstroy0