dstroy0/InputHandler #23: Naming conventions

dstroy0, 2022-03-18:
methods not intended to be exposed to users will be moved to private and will be preceded with an underscore and begin with a lowercase letter if they are not a helper method or they will start with the parent method name and end with a very brief description or mnemonic of the helper methods purpose

Protected methods will begin with a lowercase letter to differentiate them from public methods

Public methods begin with an uppercase letter

All method names are camelCase 

All class private variables are lowercase, should begin with an underscore, end with and underscore, and are underscore delimited 

I think UITYPE can be reduced to UI since it's always used like UI:: to reduce some clutter.

I don't think the argument type enum members have the best names

I think that the constants only utilized by a single class should be preceded with that classes name followed by an underscore and the constants lowercase name

I don't know if there's anything else I should add, everything is well on its way to being consistent.

Going to start on this in the next couple of days

2bndy5, 2022-03-18:
## Notes

1. Typical C++ convention dictates that all functions start with a lowercase letter. The 1 exception to this is the underscore which is a commonly used prefix to denote private members.
2. `UI` might be so short and ambiguous that it could cause name collision in other libraries.

2bndy5, 2022-03-18:
Also, many languages prefer callback aliases include `_callback` or `_cb` suffixed to the alias name. I don't often see this followed though.

dstroy0, 2022-03-18:
Ohhh because the class names start with uppercase, that makes sense.

dstroy0, 2022-03-18:
I pushed the first pass.

dstroy0, 2022-03-21:
I think this is good, let me know if there's something I missed.  If not I'll close tomorrow.

closed 2022-03-22 by dstroy0