dstroy0/InputHandler #25: prefer memcmp_P over strcmp_P

dstroy0, 2022-03-18:
memcmp_P is inherently more performant than strcmp_P, now that the process behavior is more robust I don't think there are any pitfalls to switching to memcmp_P as the command string comparator.

2bndy5, 2022-03-18:
`memcmp()` doesn't look for or stop comparing on a null terminator like `strcmp()` does. I would favor using `memcmp()` since a bytearray could have a null byte within it (thus not a string).

dstroy0, 2022-03-19:
I switched to memcmp, the only caveat is that the size variable also needs to be read from pgm.

```
if (memcmp_P(_data_pointers_[0], cmd->prm[0].command, (uint16_t)pgm_read_dword(&(cmd->prm[0].command_length))) == false) // match root command
```

closed 2022-03-19 by dstroy0