dstroy0/InputHandler #36: Examples you would like to see

dstroy0, 2022-04-03:
Any ideas for examples you would like, you can post here and we'll give it a whirl!

2bndy5, 2022-04-03:
having an example that uses a different stream is very useful. I like the current examples which cover Serial and MQTT streams. 

I think the GPS parser idea could be considered an advanced example.

dstroy0, 2022-04-04:
RF24, lora, zigbee.. any others?

dstroy0, 2022-04-04:
I think there's telnet across most platforms too so I can see if that is feasible

2bndy5, 2022-04-04:
BLE UART

dstroy0, 2022-04-04:
the teensy 4.1 has PHY ethernet and a good amount of resources, I wonder if that one can do SSH, I'll look into arduino/esp SSH while I'm looking into telnet

dstroy0, 2022-04-08:
I'm planning on starting on this once the userInput ctor overhaul is complete and wildcards have been implemented, by then the api should hopefully not be going through any more growing pains and it will make sense to work on the examples

dstroy0, 2022-04-23:
Just pushed an RF24 example, it's a repurposed GettingStarted

dstroy0, 2022-04-23:
Pushed LoRa tx/rx

dstroy0, 2022-04-24:
pushed ble uart

dstroy0, 2022-04-24:
I think I'm going to use TelnetStream as the Telnet lib.  https://github.com/JAndrassy/TelnetStream and we can add TLS with this lib: https://github.com/OPEnSLab-OSU/SSLClient

dstroy0, 2022-04-30:
debugging example is up

dstroy0, 2022-05-20:
I made a config_howto example and focused the debugging example some more

dstroy0, 2022-05-20:
I think I'm going to leave out telnet.