dstroy0/InputHandler #37: Example consolidation

dstroy0, 2022-04-03:
For most of the examples, the only real difference is the wifi lib so I think it would make sense to consolidate "universal" examples to a third folder within examples "all_platforms".  This will make maintenance easier, by reducing repetitive examples.

2bndy5, 2022-04-04:
I often see examples that cover a wider range of specific features by using 
```cpp
#ifdef ARDUINO_ARCH_ESP8266
// do stuff for WiFi
#else
// do stuff for any platform
#endif
```
I'm sure there are ways to find out if the board selected has WiFi by using `ARDUINO_ESP8266` or something like that. The macros that are used for each board are often located in the corresponding Arduino core's boards.txt. I think [the `TARGET_NANO_RP2040_CONNECT` macro would be good for the Nano Connect board](https://github.com/arduino/ArduinoCore-mbed/blob/9eeba8d63e957fec1eba81b29299d1af6d856ad4/variants/NANO_RP2040_CONNECT/defines.txt#L39). Other boards whose main processors have WiFi will be easier to identify.

dstroy0, 2022-04-04:
awesome, I'll work on those portability directives and roll those examples into all_platforms

closed 2022-04-17 by dstroy0