dstroy0/InputHandler #40: processInformation

dstroy0, 2022-04-07:
estimate ram use by adding up the different buffers that could be used, report scope timeline of them, etc.  Just more helpful information for debugging advanced implementations.

2bndy5, 2022-04-07:
You could also use [the arduino/report-size-deltas action](https://github.com/arduino/report-size-deltas) to compute mem usage and output it in a PR comment, but I think that would be more applicable for a stable state of the lib.

dstroy0, 2022-04-08:
This is an extension of some of the computing that happens in addcommand.  What I want is a per command estimate so that we can abort early if there's not enough memory instead of allocating some and then immediately have to cleanup.  

It will be precomputed so comparison will be fast.

Speaking of precomputing and compute once.  There are quite a few areas that need that now in the branch I've been working on so I'm going to open an issue for that.

That's what this is about mostly, getting the estimate for a performance increase but displaying it to assist users in planning as well.

2bndy5, 2022-04-08:
> What I want is a per command estimate so that we can abort early if there's not enough memory instead of allocating some and then immediately have to cleanup.

This sounds like a job for `malloc()`

dstroy0, 2022-04-08:
I'm already checking for nullptr at individiual alloc steps, I'm just wondering if I can precompute the size so I only have to check for nullptr once, but I don't know how safe that is?

dstroy0, 2022-05-27:
I think this is resolved

closed 2022-05-27 by dstroy0