dstroy0/InputHandler #41: Factor static recomputes to compute once

dstroy0, 2022-04-08:
Compute static equations at runtime once instead of n times, assign result to private class member.

Reduce unnecessary overhead.

dstroy0, 2022-04-18:
Most of the overhead is going to be in readCommandFromBuffer and associated helper functions.  The max number of data pointers is determined in begin(), and the type match flag array is allocated there as well instead of every iteration.

dstroy0, 2022-04-26:
I've been through all the helper functions and public methods that readCommandFromBuffer touches and condensed local parameters into objects that make sense and pass them by reference wherever possible.

dstroy0, 2022-04-26:
the maximum number of parameters passed to a function is now 5, _calcCmdMemcmpRanges is only iterated over inside of addCommand, I think that is acceptable.  Internally I tried to limit it to two or three parameters and pass by reference.  Having structs like that makes it easier to compare and see if I am duplicating something already there.

dstroy0, 2022-04-27:
Do you have any suggestions?  I think I've found everything obvious.

2bndy5, 2022-04-27:
Based on the issue title, I think you could make use of macro functions.
```c
#define rf24_max(a, b) (a > b ? a : b)
```
But, I'm not sure if this the best suggestion...
- No parameter types, so allocation is decided by the compiler
- They basically seem like `inline static` functions
- I believe compute time (compile time vs run time) depends on how the function is written (use of `const` parameter values or anything known at compile time)

disclaimer: I'm not well trained in C - I know C++ better than C.

dstroy0, 2022-04-27:
I'll look for candidates to turn into function-like macros.  Great suggestion.  If you see anything let me know.

dstroy0, 2022-04-27:
I think it's a great suggestion because we're doing a ton of repetitive things here and function like macros will probably end up being the fewest total instructions, so it will likely increase performance if we can implement them for even a few things.

dstroy0, 2022-04-27:
Like maybe for a fake iterator like Linux Kernel does?

```c

#define list_for_each_safe(pos, n, head) \
for (pos = (head)->next, n = pos->next; pos != (head); \
     pos = n, n = pos->next)

```

That might be good.

dstroy0, 2022-05-06:
If I do some iterators and it either increases speed or decreases pgm space I'll consider it a win.

dstroy0, 2022-07-23:
I'm still working on this but it's on the back burner.

closed 2022-07-23 by dstroy0