dstroy0/InputHandler #46: build-doxygen is not deploying the correct webpage

dstroy0, 2022-04-22:
I haven't even started to dig into it, just noticed it.  I checked and the build artifacts are correct, so something is up with deployment I'm guessing.

dstroy0, 2022-04-22:
`index.html` hasn't been updated for a month according to the commit history on the gh-pages branch, so I'll start looking for what happened around then.

2bndy5, 2022-04-23:
This is interesting. It seems related to [changing the workflow step to use a different action](https://github.com/dstroy0/InputHandler/commit/7f7986d38886927dd3e2031dc73316dd3437d24a).

Looking at the gh-pages branch, it seems that the doxygen output is copied to the gh-pages branch, but part of the build path remains (everything is inside the html folder). I'd guess that changing the [gh-pages settings](https://github.com/dstroy0/InputHandler/settings/pages) for this repo is the answer.
[image]
Hopefully, you can select a folder named "html" from the drop-down options.

2bndy5, 2022-04-23:
I would argue that this a problem with [the action being used](https://github.com/JamesIves/github-pages-deploy-action). If the suggestion from my last comment isn't helpful, then you'd need to play with the action's options to make docs/html get copied to the root folder of the gh-pages branch. My guess for that would be setting
```yml
      - name: Deploy 🚀
        uses: JamesIves/github-pages-deploy-action@v4.2.5
        with:
          branch: gh-pages # The branch the action should deploy to.
          folder: ./docs/html # The folder the action should deploy.
          target-folder: '.' # maybe try this??
```

dstroy0, 2022-04-23:
It was overwriting some of the files but not all of them... I just deleted the directory and ran the workflow again, and it only copied over some of the files

2bndy5, 2022-04-23:
Yeah, I noticed that too, but didn't think it was the main problem. The doxygen output _may be_ compatible with parts of previous build, but its easier to just assume it isn't. The action is assuming that previous builds _are_ compatible with newer builds 👎🏼

dstroy0, 2022-04-23:
It looks like it's something to do with `clean-exclude`, if I remove that line the redirect gets removed, and the built files get copied over to gh-pages correctly.  If that line is there it doesn't update `index.html` with the newly built file.

2bndy5, 2022-04-23:
Oh, I didn't realize the index.html in the gh-pages branch's root folder is a redirect. I've always made sure that doxygen's output (including index.html) was in the gh-pages branch's root folder...

dstroy0, 2022-04-23:
I couldn't sort out how to make it deploy into the root, so I added the redirect.

2bndy5, 2022-04-24:
> I couldn't sort out how to make it deploy into the root, so I added the redirect.

This is why I think the problem is with your choice of gh-pages deploying action. I can understand using an action that doesn't require a github token (for security risks), but maybe that action isn't the right choice with doxygen.

Alternatively, you could set doxygen's [HTML_OUTPUT](https://www.doxygen.nl/manual/config.html#cfg_html_output) to `.`, but that is relative to the [OUTPUT_DIRECTORY setting](https://www.doxygen.nl/manual/config.html#cfg_output_directory). This suggestion is talking about the doxyfile..

dstroy0, 2022-04-24:
You know way more about this stuff than I do, I'll look at other actions tomorrow and see if I can't make one work.

dstroy0, 2022-04-25:
I need to fix the Doxyfile so the mattnotmitt action will build successfully

dstroy0, 2022-04-25:
Took me a while to figure out it was a docker container and that the paths were different lol

dstroy0, 2022-04-25:
It's building correctly, deploying correctly and not overwriting the redirect.

2bndy5, 2022-04-25:
> mattnotmitt build action and peaceiris deploy action

This is a reliable combo. I had used mattnotmitt's build action for a long time until I needed specific versions of doxygen.

This may sound obvious, but have you tried removing the redirect manually?

dstroy0, 2022-04-25:
Yeah I tried that, I can make it deploy to whatever folder I want but it stays inside of the `html` directory.  I changed the destination_dir to docs, and it commits all the built files to `/docs/html/`.

2bndy5, 2022-04-25:
Strange. [We use docs/html as the output path for RF24* libs also](https://github.com/nRF24/RF24/blob/master/.github/workflows/doxygen.yml), but I haven't had problems with the path on gh-pages.

dstroy0, 2022-04-26:
Tomorrow I will try and change doxygen's output dir to root/build and try and change the actions publish dir to build/ to see if that deploy will commit to the root of gh-pages branch so we can get rid of the redirect permanently.

dstroy0, 2022-04-26:
So, the deploy action is saying that it can't find the directory that I'm building to.

2bndy5, 2022-04-26:
> error: tag OUTPUT_DIRECTORY: Output directory '/home/runner/work/InputHandler/InputHandler/build' does not exist and cannot be created

This is going to sound a little confusing, but InputHandler/InputHandler is the root of your repo after the `actions/checkout@vx` step completes. doxygen is looking for a "build" directory at the root which there is none.

In my experience, Doxygen will only create directories for the HTML_OUTPUT, it will require that the OUTPUT_DIRECTORY specified already exists.

> OUTPUT_DIRECTORY       = /home/runner/work/InputHandler/InputHandler/build

This setting is meant to be relative to the location of your Doxyfile. The Mattnotmitt action isn't actually running within the same File System that the other steps of your workflow is running in; this is because the workflow runner is a VM, and some actions (any that aren't powered by TypeScript) use Docker containers (which is like a Virtual OS) isolated from the VM with a shared directory (the env var GITHUB_WORKSPACE)... I could go on, but you have a lot to digest there.

1. If you want to use the root directory for the OUTPUT_DIRECTORY option, then just specify `..` since your Doxyfile is in your docs folder.
2. If you want the HTML_OUTPUT to go into a "build" folder at root, then specify `build` since that is relative to the OUTPUT_DIRECTORY.

closed 2022-04-26 by dstroy0