dstroy0/InputHandler #48: config file

dstroy0, 2022-05-06:
It's a mess, it needs to be split.  function like macros should be on their own.  User configurable items should be in their own file so they don't accidentally change some of our magic numbers.

2bndy5, 2022-05-06:
It might seem discouraging to embark on a refactoring crusade, but if you're at that point, then you may be right around the corner from production-ready code.

dstroy0, 2022-05-16:
Everything is split up into separate files for now, I'm going to coalesce it into three files, config, advanced_config, and noedit.

dstroy0, 2022-05-21:
I've got the framework in place mostly to swap types of certain variables automatically.  That's going into noedit inside of the config error checking section.  
Example of what I'm planning on, but I will extend it to uint32
```cpp
    #if UI_MAX_ARGS > UINT8_MAX && UI_MAX_ARGS < UINT16_MAX
        #undef num_args_group_type
        #define num_args_group_type uint16_t
    #else
        #error error message
    #endif
```

dstroy0, 2022-05-21:
I want to give users the option to disable _ui_out except for echo in

dstroy0, 2022-05-22:
hmm I don't like how this is working, because the #define is in a different translation unit it violates odr.  I think I'm going to remove the encap around the config items inside of config.h and just have users edit that directly.  Then the typedef sizing works correctly.

dstroy0, 2022-05-22:
Ok I made those changes, I still need to put a switch in for _ui_out

dstroy0, 2022-05-28:
@2bndy5 

src/config/noedit.h
```c
#if UI_MAX_COMMANDS > UINT32_MAX
        #pragma message(" at " LOCATION)
        #warning UI_MAX_ARGS cannot be greater than UINT32_MAX
#endif // end UI_MAX_COMMANDS
```

Any ideas on why changing the #warning to #error causes compilation to fail for rp2040?

2bndy5, 2022-05-28:
There are several API layers that the Arduino core goes through for the RP2040. The base layer being the actual [Pico SDK](https://github.com/raspberrypi/pico-sdk) (used for all RP2040 based boards). I'm guessing `#error` isn't defined in one of the layers (seeing the actual compilation failure message may help detirmine which layer to start with).

As a workaround, you can
```c
#if UI_MAX_COMMANDS > UINT32_MAX
        #pragma message(" at " LOCATION)
        #ifdef PICO_BUILD
                #warning UI_MAX_ARGS cannot be greater than UINT32_MAX
        #else
                #error UI_MAX_ARGS cannot be greater than UINT32_MAX
        #endif
#endif // end UI_MAX_COMMANDS
```

dstroy0, 2022-05-28:
I'll post the compiler output when I'm at the computer next.  It throws the #error but not the #warning; all you have to do reproduce is change #warning to #error in noedit.h in the error checking section for UI_MAX_COMMANDS.  #error looks to be defined but something is causing it to be thrown, I thought it might be a path issue with the #include so I prepended config/.  No obvious change.

dstroy0, 2022-05-28:
In noedit.h, changing line143 from #warning to #error
```text
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Douglas\AppData\Local\Arduino15\packages -hardware C:\Users\Douglas\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Douglas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Douglas\Documents\Arduino\libraries -fqbn=arduino:mbed_nano:nanorp2040connect -vid-pid=2341_0042 -ide-version=10819 -build-path C:\Users\Douglas\AppData\Local\Temp\arduino_build_264556 -warnings=default -build-cache C:\Users\Douglas\AppData\Local\Temp\arduino_cache_196032 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.imgtool.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.imgtool-1.8.0-arduino.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.openocd.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -verbose C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\examples\all_platforms\advanced\debugging\debugging.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Douglas\AppData\Local\Arduino15\packages -hardware C:\Users\Douglas\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Douglas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Douglas\Documents\Arduino\libraries -fqbn=arduino:mbed_nano:nanorp2040connect -vid-pid=2341_0042 -ide-version=10819 -build-path C:\Users\Douglas\AppData\Local\Temp\arduino_build_264556 -warnings=default -build-cache C:\Users\Douglas\AppData\Local\Temp\arduino_cache_196032 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.imgtool.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.imgtool-1.8.0-arduino.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.openocd.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -verbose C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\examples\all_platforms\advanced\debugging\debugging.ino
Using board 'nanorp2040connect' from platform in folder: C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\3.0.1
Using core 'arduino' from platform in folder: C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\3.0.1
Detecting libraries used...
[... cut ...]
C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\src/config/noedit.h:143:10: error: #error UI_MAX_ARGS cannot be greater than UINT32_MAX
         #error UI_MAX_ARGS cannot be greater than UINT32_MAX
          ^~~~~
Using library InputHandler at version 1.0.0 in folder: C:\Users\Douglas\Documents\Arduino\libraries\InputHandler 
exit status 1
Error compiling for board Arduino Nano RP2040 Connect.
```
It throws the error when it should not.  Changing back to a warning, the warning is not thrown.  Something is up with the 3.0.1 nano-os boards pkg.

dstroy0, 2022-05-28:
I forgot to add, UI_MAX_COMMANDS is set to 32.  The error output is incorrect, it should be UI_MAX_COMMANDS.  It's happening before pragmas are evaluated... I don't know when in the process gcc extensions are evaluated.

dstroy0, 2022-05-28:
I also had to change the syntax of the preprocessor statement for those platforms, it was `#if #elif #elif #elif #endif`.  To get it to compile I had to change it to individual statements.

dstroy0, 2022-05-28:
nano33ble, the other platform on this package:
```text
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Douglas\AppData\Local\Arduino15\packages -hardware C:\Users\Douglas\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Douglas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Douglas\Documents\Arduino\libraries -fqbn=arduino:mbed_nano:nano33ble -vid-pid=2341_0042 -ide-version=10819 -build-path C:\Users\Douglas\AppData\Local\Temp\arduino_build_264556 -warnings=default -build-cache C:\Users\Douglas\AppData\Local\Temp\arduino_cache_196032 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.imgtool.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.imgtool-1.8.0-arduino.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.openocd.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -verbose C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\examples\all_platforms\advanced\debugging\debugging.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Douglas\AppData\Local\Arduino15\packages -hardware C:\Users\Douglas\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Douglas\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Douglas\Documents\Arduino\libraries -fqbn=arduino:mbed_nano:nano33ble -vid-pid=2341_0042 -ide-version=10819 -build-path C:\Users\Douglas\AppData\Local\Temp\arduino_build_264556 -warnings=default -build-cache C:\Users\Douglas\AppData\Local\Temp\arduino_cache_196032 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dfu-util.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.imgtool.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.imgtool-1.8.0-arduino.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\imgtool\1.8.0-arduino -prefs=runtime.tools.openocd.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -verbose C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\examples\all_platforms\advanced\debugging\debugging.ino
Using board 'nano33ble' from platform in folder: C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\3.0.1
Using core 'arduino' from platform in folder: C:\Users\Douglas\AppData\Local\Arduino15\packages\arduino\hardware\mbed_nano\3.0.1
Detecting libraries used...
"C:\\Users\\Douglas\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\Douglas\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mb [... cut ...]
                 from C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\examples\all_platforms\advanced\debugging\debugging.ino:16:
C:\Users\Douglas\Documents\Arduino\libraries\InputHandler\src/config/noedit.h:148:10: error: #error UI_MAX_COMMANDS cannot be greater than UINT32_MAX
         #error UI_MAX_COMMANDS cannot be greater than UINT32_MAX
          ^~~~~
Using library InputHandler at version 1.0.0 in folder: C:\Users\Douglas\Documents\Arduino\libraries\InputHandler 
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

dstroy0, 2022-05-28:
Ok I can confirm that it's isolated to nano-os platforms.  I'm going to head over there and see if something pops out in their issues.

dstroy0, 2022-05-28:
I tested compilation on all other supported platforms.

dstroy0, 2022-05-28:
I figured out a fix:

```c

    #if UI_MAX_COMMANDS <= UINT8_MAX
        typedef uint8_t max_command_type;
    #endif    
    #if UI_MAX_COMMANDS > UINT8_MAX && UI_MAX_COMMANDS <= UINT16_MAX
        typedef uint16_t max_command_type; 
        #pragma message(" at " LOCATION)
        #warning UI_MAX_COMMANDS|max_command_type changed from uint8_t to uint16_t        
    #endif
    #if UI_MAX_COMMANDS > UINT16_MAX && UI_MAX_COMMANDS < UINT32_MAX
        typedef uint32_t max_command_type;
        #pragma message(" at " LOCATION)
        #warning UI_MAX_COMMANDS|max_command_type changed from uint8_t to uint32_t
    #endif
    #if UI_MAX_COMMANDS > ((UINT32_MAX) - 1)
        #pragma message(" at " LOCATION)
        #error UI_MAX_COMMANDS cannot be greater than UINT32_MAX
    #endif // end UI_MAX_COMMANDS

```

There must be a different limit set for the preprocessor in mbed os.

2bndy5, 2022-05-28:
I never liked the idea of using mbed os as an API layer for Arduino cores; it seemed like extra complexity to satisfy laziness.

Well, you seem to be on the right track (sounds like a problem with mbed os memory allocation).

To be of any help, I need to familiarize myself with both the code from noedit.h and the mbed os cores. Last time I looked into an Arduino core that wrapped the mbed os API into the Arduino framework API, it was rather confusing due to all the headers and the way they're organized.

dstroy0, 2022-05-28:
I thought you were plenty helpful, put me on the right track.

dstroy0, 2022-05-28:
Ok, now I'm working on that toggle for _ui_out and another one for "echo only"

dstroy0, 2022-05-28:
I pushed that out.  Just playing with the library here and there adding it into projects and changing things to see if it breaks.

dstroy0, 2022-07-23:
No more major changes to the config file anticipated.

closed 2022-07-23 by dstroy0