dstroy0/InputHandler #56: breathe .. doxygenvariable:: directive and PROGMEM issue

dstroy0, 2022-09-30:
Using the directive on valid C++11 variable declaration statements that contain the PROGMEM macro throws an invalid C++ warning because it does not find the end of the statement.  Looking for a workaround, tried moving the PROGMEM macro to after var type with same results.

2bndy5, 2022-09-30:
[I often use a predefined macro to force documentation to show](https://github.com/nRF24/RF24/blob/d7c4041f001a7fbc880535f5151f89a56a520072/Doxyfile#L374) for non-x86[_64] platforms (like Arduino).

```ini
PREDEFINED             = DOXYGEN_FORCED
```
then I can force things to show in the docs with
```cpp
#if defined(ARDUINO) || defined(DOXYGEN_FORCED)
    /** some docs */
    void func_for_arduino();
#endif
```

dstroy0, 2022-10-03:
Ok, it's not just PROGMEM it's any var with an attribute.  That tells me something is not correct in the Doxyfile probably.

2bndy5, 2022-10-03:
sounds like a preprocessing problem on behalf of doxygen. Doxygen has some detailed explanations in their manual: https://www.doxygen.nl/manual/preprocessing.html

dstroy0, 2022-10-03:
It was a few things.

1. doxygen thought it was inside of a group still at the end of the document because it picked up comments it shouldn't have.
2. using @name inside of a group freaks out doxygen.
3. the predefined syntax is very particular, and I had to have expand macros set to true.

2bndy5, 2022-10-03:
Yeah, grouping in Doxygen is without a doubt one of the worst implementations it has (second only to the archaic HTML output).

dstroy0, 2022-10-03:
It looks like I only have to keep the namespace in a group so I will go through the codebase, remove all of the groups except that one, do docs build regression testing and then close this issue if it checks out.

dstroy0, 2022-10-03:
Everything looks good, I was able to pull the namespace out and use the `@namespace` Doxygen directive, and pull it out with breathe's `doxygennamespace` directive.

closed 2022-10-03 by dstroy0