dstroy0/InputHandler #57: move constants to namespace

dstroy0, 2022-10-01:
ih_constants:: makes sense as a name to me, unless you have any ideas?

all the lib constants should go in here

2bndy5, 2022-10-01:
It may get a bit "wordy" if the namespace is repeated in the namespace's members.
```cpp
namespace InputHandler {

const char* InputHandlerConst = "";

}
```
which would be referenced as `InputHandler::InputHandlerConst`.

Otherwise, namespaces are a great habit! In python, each module is a namespace 😉

dstroy0, 2022-10-01:
yes, I would shorten the names significantly, I just need to test that it works correctly.  I did some preliminary testing to make sure that the members actually ended up in progmem.

dstroy0, 2022-10-01:
Just doing some regression testing and then I'll add the other members.

dstroy0, 2022-10-01:
Looks like there's a separate preprocessor issue for mbed platforms.

dstroy0, 2022-10-02:
constants are in their own namespace

closed 2022-10-02 by dstroy0