dstroy0/InputHandler #61: build binaries workflows

dstroy0, 2022-10-16:
Dependency adventures!  Linux is fun!

dstroy0, 2022-10-16:
Just trying to get rid of the WARNINGS in the build logs.

dstroy0, 2022-10-16:
Then I'll worry about saving the built binary.

dstroy0, 2022-10-16:
it looks like it's building with no warnings, I'll figure out how to save it to the repo and test it tomorrow

dstroy0, 2022-10-16:
Ok now I just need to figure out how to push the changes in the CI.

dstroy0, 2022-10-16:
Ok, I cloned the main branch into release and protected release so this workflow should finish correctly now, just waiting on it now.

2bndy5, 2022-10-16:
Apparently, the `set-output` command is now deprecated in GitHub action workflows. They recommend using a new variable in the github env file: https://docs.github.com/en/actions/using-workflows/workflow-commands-for-github-actions#environment-files

dstroy0, 2022-10-16:
Yeah, some external actions are using `set-output`.

2bndy5, 2022-10-16:
yeah, we plan to migrate the cpp-linter action to the env file approach soon.

dstroy0, 2022-10-16:
It looks like this is fixed.  I want to tweak the CI a bit more but this is working fine.  A build happens on push to tools/cli_gen_src that targets tools/bin/latest and one on release to tools/bin/release.

closed 2022-10-16 by dstroy0

2bndy5, 2022-10-16:
There is a way to upload build artifact from the CI to a published release's assets (which by default is just a zip and tar.gz files containing a snapshot of the repo at the tagged commit): see [csexton/release-asset-action](https://github.com/csexton/release-asset-action).

I often upload my CI build artifacts using [actions/upload-artifact](https://github.com/actions/upload-artifact)