dstroy0/InputHandler #67: reduce pgm size by consolidating string literals

dstroy0, 2023-02-12:
I will start with canned error messages.

dstroy0, 2023-02-16:
This is going well but they need to be split into their own namespace.  I'm going to try putting lib specific namespaces into their own file, and nesting the lib specific namespaces inside of the class to reduce code length.

dstroy0, 2023-02-17:
This is how I'll implement it after everything is moved around.

dstroy0, 2023-02-26:
https://github.com/dstroy0/InputHandler/blob/main/src/config/utility/namespace.h#L628

The next step is to get rid of the preallocated fixed string widths where possible to further reduce program space.  I'll be using strnlen_P, and use a buffer one char larger than that terminated with '\0' if for some reason the char array isn't already.  Then the prealloc defines become MAX instead of allocating all that space.  The most savings will come from doing this for commandString, which is 32 char fixed right now when most commands that I've been using are 8 char in length or less.

dstroy0, 2023-02-26:
I also have to add the "non-verbose" terminal output format strings and error messages.

2bndy5, 2023-02-26:
> I also have to add the "non-verbose" terminal output format strings and error messages.

I assume this isn't a reference to the [ASCII control sequences](https://en.wikipedia.org/wiki/ANSI_escape_code) because the Arduino IDE Serial Monitor doesn't support them. This might be interesting if ported to Linux, but most Linux distros ship with ncurses.