dstroy0/InputHandler #72: Nordic NRF compatibility

dstroy0, 2023-03-22:
It needs vsnprintf_P, snprintf_P, maybe pgm_read_dword.  the hook is the `NRF_H` macro.

dstroy0, 2023-03-22:
It looks like it will work, but their stream objects are significantly more complex to set up and differ platform-platform.  So unless I want to write a stream shim they'll just have to use a buffer and figure it out on their own.

2bndy5, 2023-03-23:
What arduino core/board did you use?

dstroy0, 2023-03-24:
possibly relevant: https://github.com/khoih-prog/TimerInterrupt_Generic/issues/4

2bndy5, 2023-03-24:
Probably better to use something more board-specific though:
```c
#if defined(USE_TINYUSB) && defined(ARDUINO_NRF52_ADAFRUIT)
    #include <Adafruit_TinyUSB.h>
#endif
```

dstroy0, 2023-03-24:
Awesome, I'll get to testing that!  Thanks!